FrameworkComputer/qmk_firmware #30: [Bug] Numpad module uses wrong equals

medo64, 2024-03-09:
### Describe the Bug

Framework 16 numpad module uses `KC_EQL` as equals key on numpad. This means that this key behaves same as main keyboard equals key; e.g. Shift shows '+' instead of '=' and different remappings of keyboard affect both.

The proper key code for numpad would be `KC_PEQL` (aka `KC_KP_EQUAL`) as that one is keypad specific and does not get remapped with keyboard.

### Keyboard Used

framework/numpad

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

michael-r-elp, 2024-03-23:
This won't work on Windows though, because it doesn't have a mapping for KC_PEQL, changing the mapping to it renders the button functionless.

icedream, 2024-04-24:
I'm not going to push for the PR to be merged under the given circumstances. I am not able to switch to Windows on this FW16 to test it but this sounds like something that would need a Windows-side fix, and I'm not expecting that to happen anytime soon.

IMO, there should at least be raised awareness for Linux users that this key can and probably should be remapped to the correct key code with VIA. I have no idea how to make that happen outside of documentation and pointers to documentation.

closed 2024-04-24 by medo64

michael-r-elp, 2024-04-24:
I forgot to add that using KC_EQL will also not always work depending on the keyboard layout being used, as on a german keyboard for example that would result in `´` being entered.
My solution has just been using a macro for entering `=`

I do wish all OSs would properly support `KC_PEQL`. But it seems peoples interest in proper support for windows seems basically non-existant.

whitslack, 2024-07-18:
I independently stumbled upon this problem because I use Dvorak and my numpad equals key was inputting a closing square bracket character. I reprogrammed the key to the correct scan code (`KC_PEQL`) using the web-based tool, and now it works correctly.

It's such a shame that Windblows doesn't know what to do with that scan code. Because of that, we will probably never have a correct firmware.