FrameworkComputer/qmk_firmware #39: [Feature Request]  Wiki page describing how to customize a Framework Keyboard using qmk

GreyGnome, 2024-07-06:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I wouldn't mind creating a wiki page how-to to stand up qmk on one's Framework 16, and actually modify the keymap. I think it may make sense to put it here. What do you think of this idea?

I bumped into a number of issues when customizing my keyboard and I'd like to spare future hackers the trouble.

heumann-a, 2024-07-07:
Hey, 

i am currently in the middle of porting my config to FW16. I got at least an compilable version of the stock settings available and need to check how i can keep VIA support available.

Currently i think there is a problem with the current code structure, which should be fixed first. FW uses various "_user()" functions in some Keyboards to initialize the default behavior on the keyboard level instead keymap level. This results in no available function for the user to put own code  logic in your keymaps. This may be fine for most people but after diving into QMK, these function are necessary to implement your own stuff.
After my proposal are accepted or the issue is fixed i am planning on providing a guide for it. In the meantime you could chedk my [Repo](https://github.com/heumann-a/qmp-keymaps) if that helps.

Did you document your problems? 
I think due the nature of QMK and distributed places of information i would rather create a forum post/guide which would probably get the most attention.

GreyGnome, 2024-07-07:
There is a forum post. The problem is, it's buried in the Framework forums; it's not intentional. There's a lot of updates to it as I fumbled around, trying to get things to work. That's when I thought of a wiki, because it can be kept up to date. The original forum post cannot be edited. So in the end, it's like "So what actually needs to take place, to get this to work?" It's not concise at all, it meanders through a back-and-forth conversation. In short, it's not a how-to guide. It's a journey, filled with bumps and bruises.

I didn't need a travel guide, I needed an instruction book. I wish I had it when I'd started. That's what I'm trying to create for people. I wonder if this is a good place for it.

chelzwa, 2024-10-12:
@GreyGnome A wiki guide would be extremely helpful! This was one of the first places I looked, so it seems like as good a spot as any to me.

closed 2025-02-12 by github-actions[bot]