FrameworkComputer/qmk_firmware #41: [Bug] Framework 16 RGB keyboard backlight after sleep does not work

Doridian, 2024-10-29:
### Describe the Bug

On the current build of the firmware (0.2.9), with my Framework RGB keyboard, the backlight does not re-enable successfully after sleep.

The issue can be "fixed" by removing these lines: https://github.com/FrameworkComputer/qmk_firmware/blob/fl16-v0.2.8-prep/keyboards/framework/matrix.c#L275-L279

I assume there is some re-init code missing after waking the controller back up or something?

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

flemesre, 2025-02-03:
This happens on mine as well. Re-seating the keyboard fixes it, but it is annoying to do so every time.

CDRXavier, 2026-01-02:
> I assume there is some re-init code missing after waking the controller back up or something?

Most likely. Probably identical reason with regular backlit keyboard's missing backlight when waking from sleep.